ruiqimao/keyboard-pcb-guide #3: Crystal footprint and name have changed.

itsaferbie, 2017-04-12:
As of version 4.0.6 the crystal footprint and name have changed in KiCad.

Here is an image of the crystals I think are closest:
[image]
They also have them for the TSX3225.

And the FA328 Footprint looks like this:
[image]
With the TSX3225 looking identical.

With the tutorial's information being for a older version, it could confuse people just starting out, and confused me as well when I updated.

oogieboogieboo, 2017-04-12:
I just started working on the tutorial and noticed this, too. According Epson's datasheet, pads 1 and 3 are connected to the crystal, and 2 and 4 are connected to the case, and it is recommended that pads 2 and 4 be connected to GND. 
[image]
I'm a complete newbie to all of this, but with the new footprint, pads 1 and 2 end up connected to pins 16 and 17 on the ATmega. Shouldn't that be pads 1 and 3 instead?

pgvoorhees, 2017-04-12:
@oogieboogieboo Yes they should. The problem comes in from switching the footprint. When footprints are designed, pin numbers are assigned to each pad. Which number goes where is defined by the crystal manufacturers datahseet (which is what you are seeing). 

To fix this, you also have to edit the schematic symbol, so that the pins in the symbol match the pins on the footprint which match the pins on the datahseet.

Let me know if you have trouble finding help on how to do that.

itsaferbie, 2017-04-12:
@oogieboogieboo I've been playing around with this since I got off work and found out that there is a crystal with 4 pins. And I have found out how to wire it up to make it have the pins on 1 and 3, with the GND being on 2 and 4.
[image]

And it should work with the SeikoEpson_FA238-4pin_3.2x2.5mm
[image]

I'm not sure if this is exactly what's needed since I'm new to pcb design myself.

oogieboogieboo, 2017-04-12:
@pgvoorhees I had wondered if editing the schematics and footprint would be necessary. Thank you for confirming that. I'll search around for information on how to make those changes. 
@itsaferbie I'll look through the KiCad library for that 4 pin crystal and try that, too. Thank you for pointing it out. I'm also new to this, but I would think that everything is fine if the "highlight net" tool shows that pins 1 and 3 on the crystal connected to 16 and 17 on the ATmega. 
As a sidenote, this makes me think that it would be good to get into the habit of always comparing the schematic symbols and footprints included in the libraries with the datasheets for the actual components to be used.

pgvoorhees, 2017-04-12:
> As a sidenote, this makes me think that it would be good to get into the habit of always comparing the schematic symbols and footprints included in the libraries with the datasheets for the actual components to be used.

@oogieboogieboo It is definitely a good idea. Until you build up a personal library of footprints you trust, it is good to double check everything. Let me also direct you to library services. [Snapeda.com](http://snapeda.com) has a good one that has some decent kicad support.

m47jiang, 2018-02-21:
I know this is an old post, but hopefully someone will see this. Assuming I messed up this part and connected pins 1 and 2 to pins 15 and 16, is there any way my pcb will still work? i.e is there another type of crystal I can get or do I just have to redesign this pcb with a new correct layout?

itsaferbie, 2018-02-23:
@m47jiang All you should need to do is update your schematic with the correct Crystal. Just use the image above where I show the wiring and you will have to make the correct footprint association then run the netlist. Then you'll need to delete the old one in the PCB editor view, and then read from the netlist again.

m47jiang, 2018-02-26:
Thanks @itsaferbie! I’ll probably do that for the next revision of my pcb. Assuming I already printed out a couple of them... is there any salvaging possible?

TheOGMrJoosh, 2018-05-03:
@m47jiang You will have to do a "bodge" to break the wrong traces and manually wire over to the correct pins.

This was done surprisingly often in classic consoles to fix issues identified after production :)

TheOGMrJoosh, 2018-05-03:
Apologies - just realised that i have necro'd this...

m47jiang, 2018-07-02:
@TheOGMrJoosh tried to google what a "bodge" was, unsuccessfully, good news is that you can just solder on the crystal at a 45 degree angle, as long as the pads aren't bridged it works fine :P

lf-, 2019-07-04:
@m47jiang I also did this on my PCB. The tutorial needs to be fixed to address this. Use schematic symbol Device:Crystal_GND24 and the appropriate footprint from the KiCad library or you *will* get burned like I did.

MrJoosh, 2019-07-05:
> @TheOGMrJoosh tried to google what a "bodge" was, unsuccessfully, good news is that you can just solder on the crystal at a 45 degree angle, as long as the pads aren't bridged it works fine :P

A "bodge" is a quick and dirty pseudo fix for an issue - usually involving cable ties and cloth tape in the mechanical world...

more info: [here](https://www.urbandictionary.com/define.php?term=bodge)