ruiqimao/keyboard-pcb-guide #5: After making a pcb what software can be used so your keyboard is recognized\useable?

thearkadia, 2017-06-27:
We want to print out a PCB to test with Keyboards and Macro keypads we've designed but aren't sure what software we can use to get it to be recognized by a computer and to be able to assign the keys to the characters we want.

ruiqimao, 2017-06-27:
You can use QMK firmware (link: https://github.com/qmk/qmk_firmware) to program the keyboard. I also have a separate GUI builder for the firmware at http://kb.rmao.io.

thearkadia, 2017-06-28:
@ruiqimao the link to the gui builder for the firmware is down. Any github link?

ruiqimao, 2017-06-28:
Sorry about that! I actually changed it to http://kbfirmware.com and forgot
to update you.

olimac839, 2018-02-03:
I have a question.
Once I have armed the entire keyboard, how do I program it? Do I have to use any other device to program it or can I program it directly through the USB port?

Thank you very much, I hope you can help me

VladTheImpaller, 2018-02-04:
You can program it with Atmel Flip via usb

Alien364, 2019-09-28:
I have a question, is it possible to enable N-key rollover through the GUI builder?

Xeraster, 2024-02-10:
Is it possible to come up with our own gpio pinout in qmk firmware or does it have to match a known working atmega32u4 keyboard's pinout? If changing gpio pinout is possible, how would one go about doing it? I'm no noob to this kind of thing but it's not obvious if or how this would be done with qmk.