ruiqimao/keyboard-pcb-guide #7: LED lighting

Flashduke, 2018-01-12:
I'd like to know how to connect LEDs to the board, and how to add more holes for the space bar, shift, and all of the long buttons holder thing. I look forward to hear from you soon!

ogaskell, 2020-09-23:
I may be wrong on this, but afaik the Atmega32U4 wouldn't have enough pins for an LED matrix. I struggled for enough pins on just a 40% keyboard, no LEDs.

For long buttons, when you're assigning footprints simply select the Mx_Alps_200 for a 2U key for example. The library includes footprints for 1U, 1.25U, 1.5U, 1.75U, 2U, 2.25U, 2.75U and 6.25U.

Dschorim, 2022-01-06:
Even thou you don't have enough pins for a LED matrix, you can still use LED's like the SK6805, SK6812 or WS2812B; they only require a single pin since they are daisy chained together and are supported by QMK (search for "QMK RGB" for more supported LED's)