ruiqimao/keyboard-pcb-guide #9: Idiot question about number of layers

sandric, 2018-04-16:
Hi and first of all huge thanks for hell of a work to put it all together. I build my own keyboard in past with just wiring switches with solder iron, but always had question in my head. Are mechanical keyboards pcbs manufactured in one layer? If not I was always curious how rows and column traces does not intercept each other? Only if somehow traces go under smd diodes between connections for those I have no idea. And yes, I know its clearly shown in cad software via bunch of screenshots, sorry for that, but I never used any and for me its hard to understand is it one layer or two. I recently obtained small cnc machine and want to make my own pcb for my homemade planck keyboard instead of wiring, and I dont need to place avr on pcb, those will be connected with wire, I just want to make pcb for actual diode matrix and switch mounting. I understand how to make it with double sided pcb - on one side I mill traces for rows and smd diodes, on other - traces for columns. But have no idea how (and if its even possible) to make it one layer.

Thanks and sorry for dumb question.

josuegaleas, 2018-04-16:
There are two layers, one for traces on the front and the other for traces on the back. Layers always come in pairs, and more complicated designs can have 4, 6, and beyond. Most of the MK designs I've seen are 2 layers, but others with LEDs have 4 layers.

sandric, 2018-04-16:
@josuegaleas ok, thanks, I just was thinking I completely stupid that can not understand how to make it only with one.

closed 2018-04-16 by sandric

josuegaleas, 2018-04-16:
No worries! There do exists designs for other things that can be done on one layer, but for MKs it would be very difficult to avoid any traces that don't crossover. Good luck on your project!