ruiqimao/keyboard-pcb-guide #12: Allow JSON Code from keyboard-layout-editor

KarBoy2314PL, 2018-10-06:
When someone wants to make their own layout and don't want to spend time making the pcb itself, how about using JSON from keyboard-layout-editor to make a pcb automatically?

https://github.com/ijprest/keyboard-layout-editor

Finchiedev, 2018-10-06:
Try [Kalerator](https://kalerator.clueboard.co/), it's really the only option out there (at least that I can find)

Finchiedev, 2018-10-08:
It's meant to be used with Eagle software, sorry for the confusion

Finchiedev, 2018-10-08:
I can't find anything for KiCAD

josuegaleas, 2018-10-08:
KiCad 5.0 allows you to import Eagle projects, maybe try that?

Tedfs3, 2018-10-14:
> When someone wants to make their own layout and don't want to spend time making the pcb itself, how about using JSON from keyboard-layout-editor to make a pcb automatically?
> 
> https://github.com/ijprest/keyboard-layout-editor

Generating a PCB automatically from KLE seems like a rather large request. There are too many variables in play for PCB design that are not addressed by the KLE interface. I'm just learning myself but have figure out that using the Raw Data from KLE and  and pasting it into Keyboard Firmware Builder will give you an idea how everything should be wired together.