ruiqimao/keyboard-pcb-guide #17: Anyone interested in updating this guide with an ARM chip?

sypl, 2019-08-02:
I'm interested in updating this guide to use a STM32F103 series chip. I feel more keyboards are going in this direction. More on board memory and other good stuff.

I'd also update it to use a type C USB connect and _maybe_ use footprints for hotswap sockets (probably Kailh's).

Would you, @ruiqimao, be interested in doing this with me, or would it be better that I fork and write a separate guide?

LouWii, 2019-10-23:
Did you end up writing this guide? I'd be interested.

sypl, 2019-11-15:
@LouWii Sorry, have been busy and stepped away from the keyboard stuff for a while. I'll maybe pick this up in the new year. I would need a collaborator though, so if anyone's interested, get in touch.

liftylol, 2019-11-15:
@sypl hey, i've been following this guide for ages now and built 2 boards with it but would be very interested in those things you mentionend. depending on what you would need help/assistance with i would be interested!

sypl, 2019-11-26:
@liftylol Hi there! Have you designed a PCB using the STM32F103 chip? I've looked at schematics for the chip and think this will suffice for a minimal keyboard:
[image]

Do you know if that's correct? I've used this to lay out a pcb and even got it manufactured, but couldn't get it working because I suck at soldering tiny, tiny components. But before I reorder it and get the manufacturer to place the components, I'd like to check that the schematics are correct first.

sypl, 2019-11-26:
@liftylol Do you have experience of making keyboard PCBs with the STM32103? If so could you advise on this?: 

I have this schematic, which I think is for the development board:
[image]

Also, a similar thing in slightly different format:
[Vcc-gnd.com-STM32F103C8-schematic.pdf](https://github.com/ruiqimao/keyboard-pcb-guide/files/3889579/Vcc-gnd.com-STM32F103C8-schematic.pdf)

I don't need the rows of header pins and I think we can do away with the bootloader bit (or is it absolutely needed?). If I get rid of those and wire everything up according to this schematic, then wire switched to GPIO pins like so, will we have a working PCB?:
[image]

hanya, 2019-11-30:
STM32F1 is pretty old as STM32 and it is known which has different register mapping.
USB D+ line has pull-down resistor in the schematic but recent STM32 contains this pull-down in the MCU.
STM32F0 is enough to make keyboard and can reduce BOM. If some people loves faster MCU, STMF3 would be good choice.

The schematic contains large capacitors in the power lines, they should be reduced to match to the keyboard like device.

D3 output should not connected to K4 output in the matrix.

sypl, 2019-12-07:
Thanks for the advice @hanya. I think I chose STM32F1 because I bought a development board for it. I think I'll try to do this with a STMF3. I'll try and find another schematic.

And thanks for pointint out the matrix error. Must have accidentally drawn an extra connection. Braindead.

Gondolindrim, 2020-01-06:
@sypl I'm in the middle of writing a keyboard design guide for the newer ARM processors, but the ones QMK uses most, STM32F303 and F072.

sypl, 2020-01-07:
@Gondolindrim That's awesome. What do you have so far? Would you be willing to share? 

I managed to grab the schematic of a discovery board, and I think these are the part (in red boxes) that I reckon need to be implemented for a keyboard:
[image]

Included:
- MCU, capacitors, crystals
- data line unit
- reset

Left out because I don't think they're needed:
- header pins (top-right)
- bootloader (top-middle)
- power source (middle-left)
- SWD interface
- LEDs

Would you concur with this, or is this a hopelessly naive reading of the schematic?

hanya, 2020-01-07:
@sypl 
You can have a look at the schematic of STM32F072 Discovery kit: 
https://www.st.com/content/ccc/resource/technical/document/user_manual/3b/8d/46/57/b7/a9/49/b4/DM00099401.pdf/files/DM00099401.pdf/jcr:content/translations/en.DM00099401.pdf

You do not need the following components in the schematic shown:
- R10, pull-down of the D+ line, sice it is contained in the MCU
- R9 and R11, can be removed, some recent MCU are prepared for termination
Other components can be omitted
- Y3, 32kHz crystal might not be used by keyboard
- Y2, 8MHz crystal, this can be omitted if you use crystal-less USB function by clock recovery system of STM32F072

Gondolindrim, 2020-01-07:
@sypl @hanya I have been developing ARM keyboards for quite a while now. See https://github.com/Gondolindrim/Tsuki
https://github.com/Gondolindrim/SharkPCB
https://github.com/Gondolindrim/ArcticPCB
https://github.com/Gondolindrim/KeebsPCB

sypl, 2020-01-15:
@hanya Thanks for the tips. Good to know even more components can be jettisoned. Not sure if both crystals can be got rid of. I'm not sure if I've seen a keyboard PCB without one. Of course I'm no expert and you may be right on this one.

sypl, 2020-01-15:
@Gondolindrim Oh I know who you are now! I've seen some of these PCBs before. It's great what you're doing for open source hardware.

You clearly know a lot more about PCB design than me, I'm sure you're going to write a better guide than I ever could. In fact it might make me trying to write one here a duplication of effort. Having said that I think I'll keep this open as I try to increase my own understanding of what it takes to create a keyboard PCB. I will more than likely seek your advice, if that's ok.

lstubbig, 2020-06-05:
@Gondolindrim Has there been progress on the guide? I would be very interested in giving it a look.

Gondolindrim, 2020-06-06:
@lstubbig @sypl 

I have decided to contribute to ai03's guide and we will eventually enrich it and add an ARM development guide too.

lstubbig, 2020-06-06:
I recently saw that one as well. I will check for more information on ARM now and then, thanks!

sypl, 2020-06-06:
@Gondolindrim @lstubbig I've not really been working on this, unfortunately. This is the ai03 guide, right? https://wiki.ai03.me/books/pcb-design/page/pcb-guide-part-1---preparations

It looks pretty thorough. Any timeline on the ARM dev guide?