ruiqimao/keyboard-pcb-guide #18: Keyswitch footprints change in the middle of the guide

flexd, 2019-08-29:
Hey.

When you first put the components on the PCB, the keyswitches do not have pins to put the diode in between. In the next screenshot they do. Clearly the footprint used in the tutorial earlier is not the correct one, but which one do I want to use?

flexd, 2019-08-29:
See 
[image] and 
[image]

MacroLens, 2019-09-01:
I don't really see the problem? You just have to place the diode on the opposite side of the key switch and use vias to connect them up. The footprint never changed?

What do you mean the key switches don't have pins to put the diodes between?

flexd, 2019-09-02:
I just noticed it when reading the guide. For someone new to creating PCBs like this it was confusing, I thought I missed a step or perhaps had the wrong component. I ended up using another component instead from a guide that was basically this one updated to KiCad 5. :-)

[image]

somayn, 2019-11-21:
> I just noticed it when reading the guide. For someone new to creating PCBs like this it was confusing, I thought I missed a step or perhaps had the wrong component. I ended up using another component instead from a guide that was basically this one updated to KiCad 5. :-)
> 
> [image]

@flexd could you tell me which kicad 5 guide you are talking about?

flexd, 2019-11-21:
> @flexd could you tell me which kicad 5 guide you are talking about?
https://wiki.ai03.me/books/pcb-design/page/pcb-guide-part-1---preparations

closed 2022-06-07 by flexd