ruiqimao/keyboard-pcb-guide #20: No programming instructions

jshepin, 2020-01-31:
Their is no instructions for code. also you need to add places for pins so you can program it

gustavoFreireS, 2020-03-27:
I think the idea is to program the atmega chip or using a programer (it would not be very good to embed a programmer port alongside the keyboard for a finished project since you will only want one port to serve as communication and power, not for development) at least for me the idea of the guide is covering only the hardware part.

If you want some advise on coding I suggest buy a programmer for tiny chips, like this one: https://www.aliexpress.com/i/32880288576.html (this is the most clear way for me since you will only need to solder the chip one time)

As for making computers recognizing the chip as a keyboard over usb you need to flash a firmware into it, here is a good collection (https://github.com/tmk/tmk_keyboard), you can flash one using a chip programmer (as I mentioned before) you can use an arduino as support to do this
This video explains it: (https://www.youtube.com/watch?v=RSvF7BpfqSg&t=152s) he made a PCB for it but could be done using only a breadboard)

gustavoFreireS, 2020-03-27:
this is the case where you do a more finished production keyboard but with you and to program and use the keyboard at the same usb port this guide covers it: https://medium.com/@monkeytypewritr/building-your-own-keyboard-from-scratch-bd0638c40850

closed 2020-07-07 by jshepin