ruiqimao/keyboard-pcb-guide #24: Don't see option for Preferences > Component Libraries

QueenLexi13, 2021-02-26:
I don't see the option under preferences to add component libraries. Am I missing something?

sk8erchen, 2021-02-26:
[image]
Is this what you are looking for? It should look like this, if not, you may need to consider reinstalling.

QueenLexi13, 2021-02-26:
> [image]
> Is this what you are looking for? It should look like this, if not, you may need to consider reinstalling.

Yes, that is what I see. Are one of those the same as "Add Component Libraries"?

MaetDol, 2021-07-05:
Note for guy lost in here, after opened `.sch` file then lookup `Preferences > Manage Symbol Libraries` then click folder icon at left-bottom side.

ps. also I think should rename to `Symbol` instead `Component`